OpenMixerProject/OpenX32 #1: Use-Case: X32 as DMX Console

fionera, 2025-06-19:
It would be fairly funny to use a X32 Console as DMX Input

xn--nding-jua, 2025-07-17:
I figured out how to control the surface of the fullsize-X32 with all buttons, faders displays, etc. Now I will try to implement libartnet (https://www.openlighting.org/libartnet-main) to create a DMX512-console. DMX-output is then possible using any ArtNet-Node with DMX output.

borzel, 2025-07-20:
Can you share how you did this? I want write a Reaper-Plugin to use the X32 as a DAW-Controller.

butrus, 2026-05-03:
> > Another idea is to use the MIDI-rtp or pmidi protocolls for MIDI.
> > See e.g. https://github.com/davidmoreno/rtpmidid or: https://github.com/davidmoreno/rtpmidid
> 
> please open a new ticket for that :-) so we can track it separately

You just closed a ticket where I suggested both DMX + rtpMIDI some time ago   :-P :-P :-D

(No, worry, It's OK... ;-)

butrus, 2026-05-03:
Just for the reference, I opened now #98