OpenMixerProject/OpenX32 #7: Constants for Buttons, Faders, etc.

borzel, 2025-07-21:
Hello, i played around with Version 0.0.1 of x32ctrl

I wonder, if the button values are the same for all Models. I have a X32 Compact, these are some of my first "discovered" values:

#define BTN_X32_TALK_A 256
#define BTN_X32_TALK_B 257
#define BTN_X32_MONITOR_DIM 258
#define BTN_X32_VIEW_MONITOR 259

#define BTN_X32_VIEW_USB 260

#define BTN_X32_48V 261
#define BTN_X32_PHASE_INVERT 262
#define BTN_X32_LOW_CUT 263

#define BTN_X32_ENCODER1 280
#define BTN_X32_ENCODER2 281
#define BTN_X32_ENCODER3 282
#define BTN_X32_ENCODER4 283
#define BTN_X32_ENCODER5 284
#define BTN_X32_ENCODER6 285

#define BTN_X32_HOME 286
#define BTN_X32_METERS 287
#define BTN_X32_ROUTING 288
#define BTN_X32_SETUP 289
#define BTN_X32_LIBRARY 290
#define BTN_X32_EFFECTS 291
#define BTN_X32_MUTE_GRP 292
#define BTN_X32_UTILITY 293

#define BTN_X32_UP 294
#define BTN_X32_DOWN 295
#define BTN_X32_LEFT 296
#define BTN_X32_RIGHT 297

@xn--nding-jua can you confirm, that your X32 Full has the same values? If not, we might need tt do some mapping between the X32 models.

borzel, 2025-07-22:
I'm unsure whats the best pattern for the constants:

1)  X32 first

#define X32_BOARD_MAIN 1
#define X32_BTN_TALK_A 302
#define X32_LED_BOARD_MAIN_TALK_A 0

2) the "thing" first

#define BOARD_X32_MAIN 1
#define BTN_X32_TALK_A 302
#define LED_X32_BOARD_MAIN_TALK_A 0

xn--nding-jua, 2025-07-22:
first option is better - looks cleaner.

borzel, 2025-08-17:
- [ ] add model specific translate function for buttons and leds

borzel, 2025-08-17:
the bulk of the definitions is made, closing here.

closed 2025-08-17 by borzel