OpenMixerProject/OpenX32 #8: X32 Compact: Screen shifted to left, has also some artifacts like an old analog screen

borzel, 2025-07-21:
**Hardware**
X32 Compact
Firmware 4.13

**Issue 1**
1.) the Screen is schifted to the left by some pixels

**Issue 2**
2.) there are random artifacts flying through the screen

**Reason**
Seems that the initialisation parameters are different for a X32 Compact than on a X32 Full

**Solution**
I copied the internal SSD and looked for the initalisation code. I will test with these values later (if they differ).

borzel, 2025-07-21:
something to try: https://github.com/apc-io/apc-8750/blob/master/kernel/drivers/video/wmt/devices/lcd-INNOLUX-AT070TN83.c

xn--nding-jua, 2025-07-22:
I shifted the screen by 7 pixels to the right. so this issue is resolved. The flying artifacts are another thing...

xn--nding-jua, 2025-07-31:
The artifacts (parts of the cursor) seem to be due to drawing the text to the framebuffer directly. When using SDL2 in graphics-mode the artifacts are not visible anymore. So I'm closing this issue for now...

closed 2025-07-31 by xn--nding-jua