OpenMixerProject/OpenX32 #13: Runtime Detection off X32 Model

borzel, 2025-07-24:
How can we detect on which Model we run? Would be helpfull for writing universal programs for all Models.

**Possible Solution**

Read the Paramters from SD-Card like MAC-Address, Serial and Model

xn--nding-jua, 2025-07-26:
At the moment the SD-Card is initialized by U-Boot correctly, but the Linux-Kernel is not initializing the SD-Card. It says "Clock error". As soon as we have access to the SD-Card, I know where to read the Device-Model, the MAC-Address and other things. Maybe you can try to get the SD-Card up and running? I'm not sure what the problem is...

borzel, 2025-07-26:
I know the location as well ;-)I try to get the sd-card working.Sent from Android device

borzel, 2025-07-27:
dmesg has no output "Clock error", where did you see it?

borzel, 2025-07-27:
in the Behringer firmware upgrade manual is mentioned, devices after 2023 should not downgrade to a certain firmware version.
Maybe there are some hardware changes between your and my X32, maybe the sd-card is connected differently.

I hope the sd-card is connected like on the PDK-board: https://www.nxp.com/design/design-center/development-boards-and-designs/i-mx-evaluation-and-development-boards/i-mx25-product-development-kit:IMX25PDK

I'm also on it right now

borzel, 2025-07-27:
my dmesg output

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.12.0-dirty (alex@prohorst) (arm-linux-gnueabi-gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #4 Sun Jul 27 10:57:26 CEST 2025
[    0.000000] CPU: ARM926EJ-S [41069264] revision 4 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: Behringer X32 - OpenX32
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x0000000083ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000083ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x0000000083ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Kernel command line: console=ttymxc4,115200 console=tty0 root=/dev/ram0 rw init=/init initrd=0x80810000 video=imxfb:800x480
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16384
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] Running RCU self tests
[    0.000000] Running RCU synchronous self tests
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] MXC IRQ initialized
[    0.000000] CPU identified as i.MX25, silicon rev 1.2
[    0.000000] Switching to timer-based delay loop, resolution 30ns
[    0.000006] sched_clock: 32 bits at 33MHz, resolution 30ns, wraps every 64585974256ns
[    0.000066] clocksource: mxc_timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 57481517202 ns
[    0.004384] Console: colour dummy device 80x30
[    0.004999] printk: legacy console [tty0] enabled
[    0.016566] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.017151] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.017486] ... MAX_LOCK_DEPTH:          48
[    0.017815] ... MAX_LOCKDEP_KEYS:        8192
[    0.018149] ... CLASSHASH_SIZE:          4096
[    0.018480] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.018814] ... MAX_LOCKDEP_CHAINS:      65536
[    0.019147] ... CHAINHASH_SIZE:          32768
[    0.019481]  memory used by lock dependency info: 4125 kB
[    0.019860]  memory used for stack traces: 2112 kB
[    0.020206]  per task-struct memory footprint: 1536 bytes
[    0.020725] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.50 BogoMIPS (lpj=332500)
[    0.021450] CPU: Testing write buffer coherency: ok
[    0.022761] pid_max: default: 32768 minimum: 301
[    0.025243] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.025884] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.073680] Setting up static identity map for 0x80008400 - 0x80008458
[    0.085576] Memory: 37956K/65536K available (9749K kernel code, 1280K rwdata, 3376K rodata, 268K init, 6547K bss, 26652K reserved, 0K cma-reserved)
[    0.096335] devtmpfs: initialized
[    0.179101] VFP support v0.3: not present
[    0.182447] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.183362] futex hash table entries: 256 (order: 1, 12288 bytes, linear)
[    0.185815] pinctrl core: initialized pinctrl subsystem
[    0.205681] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.221536] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.330796] imx25-pinctrl 43fac000.iomuxc: no fsl,pins and pins property in node /soc/bus@43f00000/iomuxc@43fac000/imx25-pdk/gpio2grp
[    0.331989] imx25-pinctrl 43fac000.iomuxc: no fsl,pins and pins property in node /soc/bus@43f00000/iomuxc@43fac000/imx25-pdk/gpio3grp
[    0.333051] imx25-pinctrl 43fac000.iomuxc: no fsl,pins and pins property in node /soc/bus@43f00000/iomuxc@43fac000/imx25-pdk/gpio4grp
[    0.334058] imx25-pinctrl 43fac000.iomuxc: initialized IMX pinctrl driver
[    0.515410] gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
[    0.584717] gpio gpiochip1: Static allocation of GPIO base is deprecated, use dynamic allocation.
[    0.645743] gpio gpiochip2: Static allocation of GPIO base is deprecated, use dynamic allocation.
[    0.708005] gpio gpiochip3: Static allocation of GPIO base is deprecated, use dynamic allocation.
[    0.794464] SCSI subsystem initialized
[    0.812158] usbcore: registered new interface driver usbfs
[    0.813579] usbcore: registered new interface driver hub
[... 76 lines cut ...]
[    8.875896] ALSA device list:
[    8.891958]   #0: Dummy 1
[    8.925496] Freeing unused kernel image (initmem) memory: 268K
[    8.945596] Kernel memory protection not selected by kernel config.
[    8.966235] Run /init as init process
[    8.983710]   with arguments:
[    8.983805]     /init
[    8.983885]   with environment:
[    8.983954]     HOME=/
[    8.984025]     TERM=linux
[    9.113390] usb 1-1.3: new low-speed USB device number 4 using ci_hdrc
[... cut ...]
[   10.415316] input: Logitech USB-PS/2 Optical Mouse as /devices/platform/soc/53f00000.bus/53ff4400.usb/ci_hdrc.1/usb1/1-1/1-1.4/1-1.4:1.0/0003:046D:C050.0003/input/input4
[   10.476068] hid-generic 0003:046D:C050.0003: input: USB HID v1.10 Mouse [Logitech USB-PS/2 Optical Mouse] on usb-ci_hdrc.1-1.4/input0
[   11.628818] fec 50038000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[   19.075536] platform sound: deferred probe pending: asoc-simple-card: parse error
[   57.505675] random: crng init done
```

borzel, 2025-07-27:
figured it out, why i didn't see your messageas at all. The status in the dts-file was wrong:

```
&esdhc1 {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_esdhc1>;
	bus-width = <4>;
	status = "disabled";                                     <---- should be "okay"
};
```

borzel, 2025-07-27:
im my dmesg i got

```
sdhci-esdhc-imx 53fb4000.mmc: card clock still not stable in 100us!.
```

borzel, 2025-07-27:
this ist the source: https://github.com/torvalds/linux/blob/adc218676eef25575469234709c2d87185ca223a/drivers/mmc/host/sdhci-esdhc-imx.c#L1000

```
	/* need to wait the bit 3 of the PRSSTAT to be set, make sure card clock is stable */
	ret = readl_poll_timeout(host->ioaddr + ESDHC_PRSSTAT, temp,
				(temp & ESDHC_CLOCK_STABLE), 2, 100);
	if (ret == -ETIMEDOUT)
		dev_warn(mmc_dev(host->mmc), "card clock still not stable in 100us!.\n");
```

it seesm that the ESDHC_PRSSTAT is getting not set. Normaly the clock ist created internally, so it should be stable without our interaction.

borzel, 2025-07-27:
The configuration should also be in SRAM, because the dcp-loader copies around 0x200 bytes to sram, so it can use the config values as well.

xn--nding-jua, 2025-07-31:
Function is implemented and working. The configuration will be read from the internal SD-card on boot using the init-script and stored in /etc/x32.conf.

closed 2025-07-31 by xn--nding-jua