OpenMixerProject/OpenX32 #23: Midi -> serial not working

borzel, 2025-08-03:
I send "test" into /dev/ttymxc4 and my computer receives "QSty=".
Also the boot messages are unreadable.

Maybe the clock is not set corrext?

[image]

xn--nding-jua, 2025-08-03:
No, the Baudrate of 115200 baud was working for me. Make sure, that you are using the pullup-resistor between pins 4 and 5 at the MIDI-output. Pin 2 is Ground. Pin 5 is then connected to pin 2 of the RS232-input:

[image]

borzel, 2025-08-03:
exactly what i did ... hmm...
i check again..

is ist _currently_ workging on your side with the latest commits?

borzel, 2025-08-03:
yes! you are right, i used an USB-RS232-Serial konverter, now with a TTL-Adapter ist working :-) Details ...

borzel, 2025-08-03:
Knowledge moved to Wiki -> https://github.com/xn--nding-jua/OpenX32/wiki/Serial-Debug-Interface-via-MIDI-ports